OSURoboticsClub/aerial_control #39: Fix DCM NaNs at startup

kylc, 2015-02-21:
Occasionally on the F3 platform (maybe others?) the DCM contains a bunch of NaN values. Not sure where they're coming from, but this breaks attitude estimation.

yoos, 2015-03-12:
Bad magnetometer or accelerometer vectors can mess up drift correction math. Observed on aerial v3.9 with uninitialized magnetometer.

kylc, 2015-03-12:
"Bad"?

yoos, 2015-03-12:
Zero vectors? I haven't actually looked at the numbers, so I said "bad."

kylc, 2015-03-12:
Hmmm makes sense. NaNs could be originating from trying to normalize zero vectors. Maybe the sensor health checks sitting in #42 should also check that readings are non-zero.

kylc, 2015-03-12:
Although I suppose a zero reading could still be valid. Maybe there's some kind of `READY?` register we can be checking.

yoos, 2016-04-05:
Has this been fixed and verified?