OSURoboticsClub/aerial_control #49: Recover rocket gracefully from in-flight reboot

yoos, 2015-05-21:
This really shouldn't be the case, but we've seen it happen. I currently suspect faulty power connectors. In any case, can we detect an in-flight firmware reboot and skip to the FLIGHT state? Is this too dangerous?

It may be as simple as checking for increasing altitude or an extended period of negative forward acceleration on startup. The latter risks a false positive if the board is not sitting upright when being powered up.

yoos, 2016-04-05:
Have not observed this with the aerial_v4 platform. Deprioritizing.

kylc, 2016-04-06:
I think Pixhawk implements this. They may have some documentation on how they safely detect a hot reboot condition.