OSURoboticsClub/aerial_control #50: Calculate rotational acceleration

yoos, 2015-12-01:
Due to the rotational acceleration being hard-coded to zero, previously valid PID gains for rotational position and velocity now result in instability.

kylc, 2016-04-06:
What are the acceleration estimates needed for? If we're already seeing noise on the position and velocity measurements (#51), won't the velocity derivative be far too noisy?

yoos, 2016-04-06:
We have position estimates, not measurements. Likewise, we should have acceleration estimates.

Hmm. I concede that when I added the angular acceleration controller and estimate, it was more for sake of completeness than for need. The controller gains are set to 1.0 in all units anyway, effectively making the controller a passthrough.

While working on what is currently implemented as the "rocket angular acceleration controller", I felt that the cascading PID control model puts an unnecessary constraint on the types of controllers we can implement. That is fodder for another ticket, though.

Regardless, would like to leave the angular acceleration _estimate_ in place. In the current setup, PID gains of 1,0,0 essentially turn a given controller into a passthrough.. kinda.