OSURoboticsClub/aerial_control #51: Add low-pass filters to world estimate

yoos, 2015-12-01:


kylc, 2016-04-06:
We can guess at a low pass break frequency for now, but a real solution to this requires logging the full rate IMU data. From there, we can make an informed decision on a low pass break frequency to filter out any white noise, and add notch filters to filter out any harmonics of the vehicle.

kylc, 2016-04-07:
Low pass filters added on accelerometer and gyroscope readings in f3efff4c43e2dc55a56b921159879999b7d4e919. Cutoff frequencies were guessed, so make sure to change them before testing on hardware. Waiting for testing to merge.

kylc, 2016-04-10:
@yoos ready to merge.

yoos, 2016-04-28:
Works on attitude estimator. Agree the filter should remain outside of sensor drivers.

yoos, 2016-04-28:
Title of issue is too widely scoped, as the immediate fix is for the attitude estimator only.

closed 2016-04-28 by yoos