OSURoboticsClub/aerial_control #57: Extract sensor reading, calibration, and health checks

kylc, 2016-04-13:
Every vehicle system does not need to be responsible for juggling a list of optional sensors. A common sensor reader should be created that can read all available sensors, can perform sensor calibrations, and can reason about aggregate sensor health.

kylc, 2016-04-14:
@yoos any ideas on how to pull the calibration out of the vehicle system? Check out the branch to see the new aggregate sensor interface.

My current thought is to add a `calibrateStep` method to the `Sensor` interface that would be called every controller step. Then we can add a `calibrated?` check to the interface. The vehicle system would call `calibrateStep` until all active sensors have `calibrated? = true`. This would likely be implemented as a separate state in the system state machine.

    switch(mode) {
    case CALIBRATING:
      sensors.calibrateStep();
      if (sensors.calibrated()) {
        mode = ...;
      }
      break;

    case ...:
    }

yoos, 2016-04-28:
Calibration must remove preset offsets to measure the chip's inherent bias. This also means the `Sensor`'s internal `offset` should not be modified until calibration is complete - otherwise we'd be calibrating against a moving target.